jekhokie/raspberry-noaa-v2 #967: [Error]

yvesclousier, 2024-05-20:
Good morning,
I noticed that the application does not display METEOR M2-X pictures passages at night.
The pictures are well saved and are present on the viewing page, but when you click on the thumbmail, the picture are not displayed.

Looking at the code, I think it's related to the 'daylight_pass' parameter, because when I modify daylight_pass in the DB to 1 for a satellite it works fine.

Here is an example where the daylight parameter was set to 1:
[https://clousier.com/captures/listImages?pass_id=1410](https://clousier.com/captures/listImages?pass_id=1410)

Here where is not set to 1 (by default)  :
[https://clousier.com/captures/listImages?pass_id=1407](https://clousier.com/captures/listImages?pass_id=1407)
While the image is clearly present :
[https://clousier.com/images/METEOR-M2-3-20240519-225444-321_corrected.jpg](https://clousier.com/images/METEOR-M2-3-20240519-225444-321_corrected.jpg)

Do you have an idea ?

Thanks for your reply.

yvesclousier, 2024-05-21:
Good morning,

 I'm sorry, I found out what the problem was. In the source code in the capture.php file it is expressly indicated not to display the visible Meteor images (221 and 321) during a night passage.
So you can delete or close this ticket. 
With my all apologies.

ve3elb, 2024-05-21:
Meteor only displays night images when IR is turned on. Currently Meteor IR is turned off so am not sure what you deleted or hoping to capture.

closed 2024-12-02 by MihajloPi