jekhokie/raspberry-noaa-v2 #977: [Error] Installation fails on xtrx-dkms module which is unnecessery

Szewcson, 2024-06-15:
**Describe the bug**
A clear and concise description of what the bug is.
When installing for the first time installation fails on building dkms module for xtrx (this is a mainline bug), which is unnecessary for installations using other SDR HW. Workaround is to uninstall this package using apt and re run install script (this package is probably loaded as dependency for something else).

**Server information (please complete the following information):**
 - OS: Raspberry Pi OS Bookworm
 - Python version: 3.11.2
 - SDR device: e.g. RTL-SDR v3

creas002, 2024-07-06:
@Szewcson , You are correct that xtrx is not necessary for installation.  It was installed as part of - gr-osmosdr, which has now been de-supported and removed from the dependency list. As part of that support for - gnuradio was also removed.

creas002, 2024-07-14:
Mihajlo removed support for -  gr-osmosdr & - gnuradio which will resolve the xtrx failure under this commit https://github.com/jekhokie/raspberry-noaa-v2/commit/c8ed70593410c206129cbb31689294225ccce519