jekhokie/raspberry-noaa-v2 #1002: add bottom - next/previous page in CAPTURE

i6cbi, 2024-07-08:
**Please Describe the problem to be solved**
(Replace This Text: Please present a concise description of the problem to be addressed by this feature request. Please be clear what parts of the problem are considered to be in-scope and out-of-scope.)

**(Optional): Suggest A Solution**
(Replace This Text: A concise description of your preferred solution. Things to address include:
* Details of the technical implementation
* Tradeoffs made in design decisions
* Caveats and considerations for the future

If there are multiple solutions, please present each one separately.

i6cbi, 2024-07-08:
Hi Miha. 
i don't know pull request 
i added possibilty to go next or previous page  even at the bottom of the page without to go back to the top
 
/var/www/wx-new/App/Views/Captures/index.html
line 102
{% endfor %}
<link rel="stylesheet" type="text/css" href="/assets/css/pagination.css">
<nav aria-label="page" id="pagination" class="mb-0">
  {% include('Captures/pagination.html') %}
{% endblock %}

creas002, 2024-07-14:
New feature request is now included in aarch-support in this commit. https://github.com/jekhokie/raspberry-noaa-v2/commit/370013bec1682a69eb491b64ba8314709c5e492f

closed 2024-07-14 by i6cbi