jekhokie/raspberry-noaa-v2 #1008: [Error] Installing SatDump on Raspberry

i6cbi, 2024-07-10:
`TASK` [common : Installing predict on Raspberry Pi (armhf)] *********************
ok: [localhost]

TASK [common : Installing predict on Raspberry Pi (arm64)] *********************
skipping: [localhost]

TASK [common : Installing SatDump] *********************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Unable to install package: E:Impossibile aprire il file /home/pi/raspberry-noaa-v2/software/satdump_1.2.1-d5fd2f41__armhf.deb - open (2: File o directory non esistente)"}

PLAY RECAP *********************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1    skipped=4    rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above
``
N.B .  File  exist

creas002, 2024-07-10:
I see an extra space added to the file name ansible is looking for

`/home/pi/raspberry-noaa-v2/software/satdump_1.2.1-d5fd2f41**_**_armhf.deb`

It should look like this 

`/home/pi/raspberry-noaa-v2/software/satdump_1.2.1-d5fd2f41_armhf.deb`

The dependencies.yml file I'm looking at in aarch64-support  does not have that space, so has your local file been modified?

$HOME/raspberry-noaa-v2/ansible/roles/common/tasks/dependencies.yml

MihajloPi, 2024-07-10:
What OS are you using? To me it looks like it's 32 bit Raspberry OS for Raspberry Pi, but which version?

i6cbi, 2024-07-10:
> What OS are you using? To me it looks like it's 32 bit Raspberry OS for Raspberry Pi, but which version?
always the same version

description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

in software i have 
 [ satdump_1.2.1-d5fd2f41_bullseye_armhf.deb](https://github.com/MihajloPi/raspberry-noaa-v2/blob/master/software/satdump_1.2.1-d5fd2f41_bullseye_armhf.deb ]

i solved .. but others ??

i6cbi, 2024-07-10:
> I see an extra space added to the file name ansible is looking for
> `/home/pi/raspberry-noaa-v2/software/satdump_1.2.1-d5fd2f41**_**_armhf.deb`
T
hanks Richard .. you are right --  solved myself .. but other users ??

creas002, 2024-07-10:
I was asking if you had a modified dependancies.yml file with the space. You confirmed you did. The source Repo that you just posted has the correct data and would not have the extra space. So I guess its a mystery?

[](https://github.com/MihajloPi/raspberry-noaa-v2/blob/master/ansible/roles/common/tasks/dependencies.yml)

Lines 196-201 is what drives ansiable to use the file name it reported in your error. You can see it does not have the extra space.

Perhaps Mihajlo may have an idea how your local file picked up the space.
```
- name: Installing SatDump
  become: yes
  apt:
    deb: "{{ noaa_home }}/software/satdump_1.2.1-d5fd2f41_{{ raspbian_version.stdout }}_{{ system_architecture }}.deb"
    state: present
    update_cache: no
```

creas002, 2024-07-10:
[](https://github.com/MihajloPi/raspberry-noaa-v2/blob/master/ansible/roles/common/tasks/dependencies.yml)

i6cbi, 2024-07-11:
> I was asking if you had a modified dependancies.yml file with the space. You confirmed you did. The source Repo that you just posted has the correct data and would not have the extra space. So I [...]

yes its same .. no modified .. when i have time .. i'll try with new  fresh istallation

creas002, 2024-07-14:
Issue was duplicated and resolved via this commit in arrach64-support  [https://github.com/jekhokie/raspberry-noaa-v2/commit/f58f1b736fa6c8182ec013ff1be5efe0d5c426c5](url)

closed 2024-07-14 by i6cbi