jekhokie/raspberry-noaa-v2 #1016: [Error] Bias T only appears to switch on for one pass after system restart

ShianMull, 2024-07-18:
I’ve recently updated to the latest code to include the Meteor M2-x satellites.  I de-installed the old version of Raspberry NOAA and installed the new version, with apparent success.

I revised the Config file to include my preferred settings, which includes enabling the 5V Bias T setting as I use an RTL-SDR V3 with compatible RTL-SDR amplifier (both RTL-SDR Blog kit).  Note that all worked very well in the previous version of software and I reliably received NOAA images extending from southern Europe to top of Greenland etc images.

However, with the new version, I seem to be able to receive one pass that works well, but all subsequent passes have very poor reception and it looks like the Bias T powering the RTL-SDR amplifier may have been turned off after the first pass, but does not turn on for subsequent passes until I reboot the RPi - I can’t see the green light turn on the amplifier.  Passes after the first one (for NOAA) barely cover the UK (and are noisy), whilst subsequent Meteor passes do not process at all.

I’m a bit stumped as to what might be happening here?

**Support Output**
Paste the output of running the `./support.sh` script here.

=============================================
Details about environment
=============================================
Current local date/time: Thu 18 Jul 09:32:54 BST 2024
Current date/time (ms):  1721291574
Repo git hash:           ef65cfac5c87fc2d44a3f952f581f28c7dafcc3c
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.4
Architecture:            Linux RaspiNOAA 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [10510	Fri Jul 19 21:48:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240719-204800 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721422080 920 76 Northbound W
  * [10428	Mon Jul 22 10:25:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240722-092543 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721640343 924 40 Southbound W
  * [10541	Thu Jul 18 16:47:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240718-154746 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721317666 906 25 Northbound W
  * [10443	Fri Jul 19 11:34:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240719-103437 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721385277 933 28 Southbound E
  * [10542	Fri Jul 19 04:55:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240719-035522 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721361322 947 73 Southbound E
  * [10479	Sat Jul 20 21:52:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240720-205256 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721508776 1008 77 Northbound W
  * [10426	Sun Jul 21 20:39:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240721-193943 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721590783 939 39 Northbound W
  * [10419	Sat Jul 20 09:37:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240720-083721 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721464641 936 81 Southbound W
  * [10446	Fri Jul 19 23:08:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240719-220847 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721426927 993 85 Northbound W
  * [10422	Sat Jul 20 21:06:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240720-200602 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721505962 857 26 Northbound W
  * [10574	Tue Jul 23 23:59:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240723-225939 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721775579 961 40 Northbound W
  * [10549	Sat Jul 20 07:54:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240720-065429 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721458469 720 11 Southbound W
  * [10556	Sun Jul 21 14:03:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240721-130315 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721566995 868 28 Northbound E
  * [10568	Tue Jul 23 16:40:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240723-154024 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721749224 892 27 Northbound W
  * [10515	Sat Jul 20 21:24:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240720-202455 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721507095 972 75 Northbound E
  * [10523	Mon Jul 22 10:49:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240722-094935 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721641775 949 55 Southbound E
  * [10486	Mon Jul 22 11:34:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240722-103444 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721644484 988 87 Southbound W
  * [10488	Mon Jul 22 19:49:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240722-184921 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721674161 849 21 Northbound E
  * [10585	Wed Jul 24 19:54:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240724-185404 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721847244 816 22 Northbound E
  * [10431	Tue Jul 23 08:20:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240723-072005 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721719205 875 30 Southbound E
  * [10584	Wed Jul 24 13:23:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240724-122338 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721823818 816 18 Southbound W
  * [10564	Tue Jul 23 05:09:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240723-040939 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721707779 965 89 Southbound W
  * [10416	Fri Jul 19 18:12:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240719-171232 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721409152 862 24 Northbound E
  * [10462	Tue Jul 23 14:05:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240723-130553 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721739953 950 33 Southbound W
  * [10430	Mon Jul 22 20:13:00 2024 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240722-191338 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1721675618 953 59 Northbound W
[... cut ...]
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

ShianMull, 2024-07-25:
It would also appear that auto gain does not work in this latest version either.  Worked fine in previous versions of the code, so not sure what has changed?

creas002, 2024-07-25:
There was a short time that there was version had an issue with gain parameter being passed correctly.   Can you report the output of having run the following command please? 

`cat $HOME/raspberry-noaa-v2/.git/config`

Would you be interested in Beta Testing the next Raspberry-NOAA-V2 which includes 64-bit support?

OS requirements : Debian 11 or 12 (32-bit or 64-bit) 

Please follow these steps to ensure best outcome. 

Installation steps:
-------------------------------------------------------
1) Perform a fresh OS install 

2) Install the latest OS updates.

   `sudo apt update`
   `sudo apt -y full-upgrade`
   `sudo reboot`

3) Use github for pulling the RN2 bookworm-support branch down. 

   `cd $HOME`
  `git clone --depth 1  https://github.com/jekhokie/raspberry-noaa-v2.git -b aarch64-support`

4) Modify settings.yml as you normally would 
    (There is a new parameter called enable_coronal)

   `vi $HOME/raspberry-noaa-v2/config/settings.yml`

5) Execute the install script 

   `cd $HOME/raspberry-noaa-v2`
   `./install_and_upgrade.sh`

4) New feature - Execute the Verification Tool and look for any unacceptable failures

   `$HOME/raspberry-noaa-v2/scripts/tools/verification_tool/verification.sh quick`

**NOTE** * If you need to remove RN2, execute the uninstall script

   `$HOME/raspberry-noaa-v2/scripts/tools/caution_uninstall_rn2.sh`

ShianMull, 2024-07-25:
Output from cat command is:

[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/jekhokie/raspberry-noaa-v2.git
	fetch = +refs/heads/master:refs/remotes/origin/master
[branch "master"]
	remote = origin
	merge = refs/heads/master

I’ll certainly have a look at the Beta testing the latest.  What typical aspects should I be aware of for this beta version?

Many thanks.

creas002, 2024-07-25:
> I’ll certainly have a look at the Beta testing the latest. What typical aspects should I be aware of for this beta version?

I'm not sure I understand the question. There are a number of incremental fixes in this version that are not in the MASTER version you are presently running, so our hopes is a better experience for users as well as supporting 64-bit OS.

ShianMull, 2024-07-25:
I was just asking if there was anything in the Beta version you would specifically like tested, but I’m guessing by your response, you’re just seeing if it works on my system.

Can I ask if the ./install_and_upgrade.sh sets a differing system username and password, because I can’t see to logon via VNC now I’ve installed if afresh?

Many thanks

creas002, 2024-07-25:
You are welcome.  Thank you for testing and happy to read its seems working for you.

Regards

ShianMull, 2024-07-26:
All seems good today as well.  May I ask if there is code in any of the branches for the original V2 Home Screen that displays processor usage, sunspot activity, etc?  I’d love to keep that going if possible.

creas002, 2024-07-26:
> All seems good today as well. May I ask if there is code in any of the branches for the original V2 Home Screen that displays processor usage, sunspot activity, etc? I’d love to keep that going if possible.

RN2 is a stripped down version which only install the things needed to receive and process APT telemetry.  I am going to guess that you are referring to the utilities that someone else is bundling together in the pre-built images that can be installed on a Micro SDCard to get you going quickly.  I believe the README references those installation paths under the heading of 

> "Super Easy setup: Use a maintained image"

 , but its my understanding that those images lag the development and code fixes the we develop here.

ShianMull, 2024-07-26:
Ok, no worries.  Many thanks for your help.

closed 2024-07-26 by ShianMull

creas002, 2024-07-26:
Absolutely. I'm pleased that things are working correctly for you.. Regards