jekhokie/raspberry-noaa-v2 #1054: [Error]"Unable to install package: E:Could not open file /home/pi/raspberry-noaa-v2/software/satdump_1.2.1-1021e84b_bookworm_arm64.deb - open (2: No such file or directory)"}

KE8AFF, 2024-08-22:
pi@raspberrypi:~/raspberry-noaa-v2 $ ./install_and_upgrade.sh
 * Installing yaml and jsonschema Python modules...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-yaml is already the newest version (6.0-3+b2).
python3-jsonschema is already the newest version (4.10.3-1).
The following packages were automatically installed and are no longer required:
  libairspyhf1 libglfw3 libjemalloc2 libmbedtls14 libmbedx509-1 libnng1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 * Checking configuration files...
SUCCESS: The configuration config/settings.yml conforms to the expected schema!
 âœ“   Config check complete!
 * Configure ATRM rule and PHP Controller based on scheduling user...
   skipping, ATRM Rule is already set...
   skipping, AdminController.php : sudo pi ATRM is already set...
 * Running Ansible to install and/or update your raspberry-noaa-v2...
[DEPRECATION WARNING]: "include" is deprecated, use include_tasks/import_tasks instead. See https://docs.ansible.com/ansible-core/2.14/user_guide/playbooks_reuse_includes.html for details. This feature
will be removed in version 2.16. Deprecation warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.

PLAY [install raspberry-noaa-v2 core] *********************************************************************************************************************************************************************

TASK [get os release] *************************************************************************************************************************************************************************************
ok: [localhost]

TASK [common : Installing general dependencies] ***********************************************************************************************************************************************************
ok: [localhost]

TASK [common : Install required Python packages using pip3 since they aren't available among the apt packages] ********************************************************************************************
changed: [localhost]

TASK [common : Install required Python packages using pip3] ***********************************************************************************************************************************************
skipping: [localhost]

TASK [common : Installing predict on PCs] *****************************************************************************************************************************************************************
skipping: [localhost]

TASK [common : Installing predict on Raspberry Pi] ********************************************************************************************************************************************************
skipping: [localhost]

TASK [common : Installing SatDump] ************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Unable to install package: E:Could not open file /home/pi/raspberry-noaa-v2/software/satdump_1.2.1-1021e84b_bookworm_arm64.deb - open (2: No such file or directory)"}

PLAY RECAP ************************************************************************************************************************************************************************************************
localhost                  : ok=3    changed=1    unreachable=0    failed=1    skipped=3    rescued=0    ignored=0

Error:   Something failed with the install - please inspect the logs above
pi@raspberrypi:~/raspberry-noaa-v2 $

creas002, 2024-08-22:
I can tell from the failure that satdump_1.2.1-1021e84b_bookworm_arm64.deb  version is what the MASTER repository tries to install. Bookworm/Debian 12 (32-bit) is recommended for the master repository per the README,

I would recommend that you try the beta-development branch as it supports both 32/64-bit Bookworm Debian 11 and Debian 12 and it has many fixes and improvements

Please **follow these steps** to ensure best outcome. 

**Installation steps:**
1) Perform a fresh OS install of Debian 12 64-bit (Full Desktop or Lite version depending on your preference)
2) Install the latest OS updates.

```
   sudo apt update
   sudo apt -y full-upgrade
   sudo reboot
```
3) Use github for pulling the RN2 beta-development branch down. 

```
  cd ${HOME}
  git clone --depth 1  https://github.com/jekhokie/raspberry-noaa-v2.git -b beta-development
```

4) Modify settings.yml as you normally would using nano or vi 

   `vi ${HOME}/raspberry-noaa-v2/config/settings.yml`

5) Execute the install script 

```
  cd ${HOME}/raspberry-noaa-v2
   ./install_and_upgrade.sh
```

KE8AFF, 2024-08-29:
I have followed the above instructions on a Raspberry Pi 4 8gb and have not been able to get any decodes. I noticed an error during the install but it appears that it was ignored. The error is as follows: 

TASK [common : Check if systemd-timesyncd service exists] ****************************************************************************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["systemctl", "status", "systemd-timesyncd"], "delta": "0:00:00.020586", "end": "2024-08-28 23:52:59.742082", "msg": "non-zero return code", "rc": 4, "start": "2024-08-28 23:52:59.721496", "stderr": "Unit systemd-timesyncd.service could not be found.", "stderr_lines": ["Unit systemd-timesyncd.service could not be found."], "stdout": "", "stdout_lines": []}
...ignoring

KE8AFF, 2024-08-30:
Double Checked the RTL Sticks and the entire antenna system and all seem to be working properly. I have a feeling that the above error is causing pass recording to fail.